alexmohr/open-heat #14: BME280 Vin Pin

Anycubic, 2021-10-30:
Good morning, I see in the diagram BME280 Vin pin is wired to GPIO15 (I guess in order not to always feed the sensor so to save the battery). BUT in the  readme also open window led is wired to GPIO15 by default.
I also tried to connect BME280 Vin to GPIO15 (I didn't configure it for window led) but it seems to me the BME is not working in this way.
Could you help me?
Thanks a lot
Giuseppe

alexmohr, 2021-10-30:
The readme is outdated and in the deep sleep branch the window sensor is not functional anymore. 
I'm concentrating on core functionality before re-adding support for this

Anycubic, 2021-10-31:
Ok thanks. But how do I enable GPIO13 for BME280? Is it implemented already or it's just in the diagram for the time being?

alexmohr, 2021-10-31:
It is implemented and working for me. You have to configure the pin on the Webinterface as well.

Anycubic, 2021-10-31:
Ok, now I understood why there was a new field in the configuration interface, I guess I was sleeping and didn't catch its purpose :-D

closed 2021-10-31 by Anycubic