alexmohr/open-heat #15: Valve "off" but valve was not closed

Anycubic, 2021-10-31:
I don't know if you saw this behaviour before, valve was set to "off" in a proper way (checked also by sniffing the broker) but valve was actually not closed. Unplugged the ESP, replugged and I heard the motor spinning for quite a few seconds and actually closing the valve.
I'm not feeding the ESP with batteries, so low voltage is out of the question.

alexmohr, 2021-10-31:
Yes I also noticed this today. 
Expect things to break because I don't have any tests whatsoever in place to prevent this.

Fixed in #d109518

Anycubic, 2021-11-02:
It seems I still have this issue, valve is closed but temperature still increasing (and radiator still very hot to touch).
I shut off the ESP, waited one minute and shut on again and the valve started closing properly eventually.

[image]

Anycubic, 2021-11-02:
Maybe you need to implement a sort of "calibration" routine to find the correct `VALVE_FULL_ROTATE_TIME` for each and different valve mechanic (if I understood well your code of course)

alexmohr, 2021-11-02:
Probably it would be best to use the ADC to check the power consumption of the valve. If it goes above a certain threshold it is at the end position. 
It should be possible with a simple voltage divider to get the voltage input to 0..1V. 
I already did some experiments with that but couldn't get it to work reliably. 
With this change the regulator algorithm could also be changed to a PID controller instead of the currently used I-Controller. 
Integral regulators tend to oscillate around the set point a bit until they stabilize. 
I don't think the change of the constant will solve the issue with overshooting.

Anycubic, 2021-11-02:
Overshooting is now much better, what you can see in my image is valve not being fully closed though it was supposed to be closed that's why you can see "overshooting". I'm going to test latest commit

Anycubic, 2021-11-05:
So I've tested latest build this days and still valve not full closing in off position, unplug -> replug and I hear for a couple of seconds the valve full closing. Do you advice increasing `VALVE_FULL_ROTATE_TIME` until I don't see the issue anymore? Thank you

alexmohr, 2021-11-05:
Try increasing it to 60s

alexmohr, 2021-11-05:
Reopening this issue as it's still not fixed for you. 
set `static constexpr int VALVE_FULL_ROTATE_TIME = 40'000;` to `static constexpr int VALVE_FULL_ROTATE_TIME = 60'000;`
I'll test this as well.

Anycubic, 2021-11-05:
Testing right now

commit d1095181 (2021-10-31): regulator: fix turn off bug

fixes #15
closed 2021-11-10 by alexmohr