alexmohr/open-heat #18: Ability to shut on/off the valve with a physical switch on the valve case itself

Anycubic, 2021-11-04:
How would be hard to implement that? Maybe using a digital input (closed/open) and writing on the proper MQTT channel could be feasible?

alexmohr, 2021-11-04:
It's not trivial with the current setup. I don't really get how you imagine MQTT being useful here. 
1. You have to wake the device from deep sleep by pulling the reset pin to 3.3V. Could be done by a button.
2. Detect somehow that the device was woken up by the button press. You can't use the same button as you have to let go to boot the device (reset = low) using a second button is also stupid because you would have to remember to order and timing would matter (a lot) 
3. You can't do anything from the network because you are in deep sleep most of the time.

Really the only way I can imagine this working is with additional hardware, like a shift register.

Anycubic, 2021-11-05:
What I was thinking is sort of simulating pressing "off" or "heat" on home assistant interface. One button, you press on "0" and the valve itself write "off" on TOPIC/mode/set, press the same button on "1" and this time "heat" is pushed on TOPIC/mode/set. I know, maybe it's a stupid idea :-D

alexmohr, 2021-11-05:
I'm not going to implement something like this. 
If you set the button to heat how could you turn heating off from homeassistant?
You could only react on changes and use the button as a toggle but then the position of the button would not indicate the state of the valve and you'd have to look into home assistant anyways. 
Besides with MQTT you still have the issue that the device needs to wake up from deep sleep on itself because you can't reset it.

closed 2021-11-05 by alexmohr