alexmohr/open-heat #20: Checking if furnace is actually open/closed

Anycubic, 2021-11-10:
It could happen that the furnace is switched off and the valve will try to open and open but of course temperature will never increase. This could also lead to servo motor damage. What about putting a check i.e. if in 30 minutes from "heat" status temperature didn't raise by 0.3 celsius then switch off the valve?

alexmohr, 2021-11-10:
Sorry but that's not in scope of this project. After 90s seconds further rotation of the motor is stopped anyways, so the danger of damaging something is not greater than with fully closing the valve.
If you need this functionality you can automate this in home assistant or by writing a simple scripts which listens to mqtt. 
If you try to implement this yourself keep in mind that you also could be in range of the set temperature already and in this case switching off heating would not be correct.

I won't accept a pull request for this feature.

closed 2021-11-10 by alexmohr

Anycubic, 2021-11-10:
No worries, I'll try to avoid unexpected scenarios with automation i.e. if thermostat is switched off then all valves switched off as well