alexmohr/open-heat #23: Please support BMP280

Anycubic, 2021-11-19:
Hello,
I just got 15 BMP280 unstead of BME280...coming from China so I guess I will keep them. Could you please support BMP280 as well? Thank you

alexmohr, 2021-11-19:
Should be fairly easy to implement. I see what I can do. But as I don't have any bmp testing will be up to you.

Anycubic, 2021-11-19:
Of course I will test it! Should be same like BME but it doesn't have humidity sensor (which I really don't care). Id is 0x76, I guess you need to declare BMP280 instead of BME280 and take the humidity code out. I made it work with ESPHome no problem, but putting my hand in your code guts is another story :-D

alexmohr, 2021-11-20:
I do care about humidity but it's not a problem to make it configurable

Anycubic, 2021-11-20:
Tested and not working (sensor not recognised). I'm sure both the sensor and the connections are ok because with ESPHome code is working fine.
I selected BMP280 but is it ok he is trying to setup the BME280 instead?

```
[21:49:56][INFO]  [0000000096] Setting up BME280
[21:49:56][INFO]  [0000000101] BMSensor init result: 0, try: 1
[21:49:56][INFO]  [0000000206] BMSensor init result: 0, try: 2
[21:49:56][INFO]  [0000000307] BMSensor init result: 0, try: 3
[21:49:56][INFO]  [0000000408] BMSensor init result: 0, try: 4
[21:49:56][INFO]  [0000000509] BMSensor init result: 0, try: 5
[21:49:59][ERROR] [0000003112] Failed to init temp sensor
```

alexmohr, 2021-11-20:
I forgot to update the configuration page. It should be working now from the initial configuration and from the normal web ui.

Anycubic, 2021-11-21:
Yes! It's working now! Thank you!

closed 2021-11-21 by Anycubic