alexmohr/open-heat #24: 18650 shield

Anycubic, 2021-11-20:
Regarding the 18650 shield. I don't know which version you are using, I got 10 pieces of the V8 version which has a push button instead of the on/off switch. This is a real problem because when in deep sleep the ESP-12F drain so low power that the shield doesn't detect it and just shut down... do you know if the there is a way to implement the good old "on/off" switch? Is it maybe the J4 in the pic? I was not able to find anything on the net.
I was also thinking about to feed directly the AMS1117 pin with Vbat like in the pic, so bypassing the shield completely or at least partially.
[image]
[image]

Anycubic, 2021-11-20:
Any suggestion? Unfortunately my 18650 right now are useless

alexmohr, 2021-11-20:
I'm using the V8 as well but my PCB is looking completely different. It also has a push button but that is only used to check the battery status. 
But I'm also not using the usb output and soldered a connector to ground and bat+
(Btw your email really hasn't been received by my Mailserver ;) )

[image]
[image]

Anycubic, 2021-11-20:
But so are you using a step down to convert to 3.3v? This will be also my last thing to do!
Do you think that plugging directly Vbat and Vin will destroy the shield? In this why I could use the internal 3.3 step down without another piece

Anycubic, 2021-11-20:
>  (Btw your email really hasn't been received by my Mailserver ;) )
> 

It seems you have a problem with your mail server?
[image]
[image]

alexmohr, 2021-11-20:
Cool thanks for the screenshot, it seems like some dns servers did not like my mx record. Should be fixed now. 
https://dnschecker.org/mx-lookup.php?query=mohr.io&dns=google

I don't think it's bad for the shield I'm using it like this for a while now.
The voltage regulator is not used as step down but because I want to prevent too large voltages at the ESP. If the battery is full it's around 4.2V which still should be fine but better safe than sorry. 
Furthermore I'm powering the ESP from an external power supply from time to time which can deliver a lot more than 4V. 
It's just a precaution and not really necessary. 
I'm also using a nodemcu v2 without the voltage regulator.

Maybe the battery runtime could be improved by not using the regulator.
The circuit for the stop detection will need a stable supply voltage and then the regulator will be necessary.

Anycubic, 2021-11-20:
Well I'm using only ESP-12F now and anything > 3.6V should make the magic smoke out of it! :-D
So I will try to feed VBat directly to the 1117 Vin pin and see what happens. Actually there must be a diode of some kind preventing that voltage going back to the circuit, unless I switch one the module with the push button, which of course I'm not going to do

Anycubic, 2021-11-20:
> I don't think it's bad for the shield I'm using it like this for a while now. The voltage regulator is not used as step down but 

For "shield" I mean the 18650 board, that's how they call it ;-)

Anycubic, 2021-11-20:
This is my 18650 board model: https://www.youtube.com/watch?v=LRWqA2Zy9Zo&ab_channel=EasyOne
AMS1117 is taking Vin from the TP5602 Vout (it's exactly 5.00V). I wonder if feeding it with the Vbat could harm it

alexmohr, 2021-11-20:
I'm sorry I never really bothered with the details of the battery shield, so I can't really help you with this. 
but if you figure things out i'd be really interested in the battery lifetime you can achieve

Anycubic, 2021-11-20:
Uhmmm, I was thinking I could just desolder and isolate the AMS1117 Vin pin from the board and feeding it directly with Vbat. This should make no harm at all. In this way the main board circuit should dissipate close to zero because TP5602 will be not powered at all. Voltage should go just to AMS1117 and those 2 capacitors close to it.
What to think, Herr Professor? :-D

alexmohr, 2021-11-20:
From your description it _should_ work but it's really hard to say for sure without schematics for me as I'm more a software guy to be honest :laughing:

Anycubic, 2021-11-20:
I'll give it a try, should I fry everything I will use just the battery holder with a step down converter

Anycubic, 2021-11-20:
It seems working good!
3.26V on the 3 rails, nothing on 5
[image]

alexmohr, 2021-11-20:
That's awesome :+1:

Anycubic, 2021-11-20:
BTW I would not advice to buy this battery shield, there are so many versions that you never know which one you will get

alexmohr, 2021-11-20:
That's a good argument. I just ordered a couple and that's why I put it in the readme. I never would have thought anyone but me would use this.
Do you have a recommendation on what to use?

Anycubic, 2021-11-21:
Given that those 18650 boards don't work with a deep sleep ESP I'go with a bare double battery holder + battery discharge protector + step down converter like these:
https://it.aliexpress.com/item/32810810604.html?spm=a2g0s.9042311.0.0.58074c4dzYiDaH
https://it.aliexpress.com/item/32920085100.html?spm=a2g0s.9042311.0.0.27424c4dPGhFEt

alexmohr, 2021-11-21:
That's a great idea. I'll order some with a battery holder and try it out.

Anycubic, 2021-11-22:
I just installed my first ESP-12F + Valve + BPM280 + modded 18650 battery shield and everything happily working together ;-)

closed 2021-11-22 by Anycubic

alexmohr, 2021-11-23:
Awesome I really curious about the battery runtime you can achieve

Anycubic, 2021-11-23:
I have the same curiosity :-D