alexmohr/open-heat #26: Get set temperature shows nothing

alexmohr, 2021-11-21:
Have you ever had this issue? Get set temperature shows nothing. I had to restart HA and it happened.
Tried to restart MQTT, HA, nothing. Deleted and pasted.again the mqtt section in HA config, had it reappeared, restarted HA and disappeared again!
[image]

_Originally posted by @Anycubic in https://github.com/alexmohr/open-heat/issues/22#issuecomment-974792905_

alexmohr, 2021-11-21:
Do you receive any mqtt message from the esp? Can you post the logs?
It's always hard to troubleshoot without logs

Anycubic, 2021-11-21:
That's what I get every ESP wakeup:

```
Message 21 received on papa/log at 14:45:
[DEBUG] [0000917634] MQTT send '4.13' in topic 'papa/battery/voltage'
QoS: 0 - Retain: false
Message 20 received on papa/log at 14:45:
[DEBUG] [0000917632] MQTT send '93.37' in topic 'papa/battery/percent'
QoS: 0 - Retain: false
Message 19 received on papa/log at 14:45:
[DEBUG] [0000917591] MQTT send '300000' in topic 'papa/modemsleep/get'
QoS: 0 - Retain: false
Message 18 received on papa/log at 14:45:
[DEBUG] [0000917589] MQTT send '51.73' in topic 'papa/humidity/measured/get'
QoS: 0 - Retain: false
Message 17 received on papa/log at 14:45:
[DEBUG] [0000917583] MQTT send '19.84' in topic 'papa/temperature/measured/get'
QoS: 0 - Retain: false
Message 16 received on papa/battery/voltage at 14:45:
4.13
QoS: 0 - Retain: false
Message 15 received on papa/battery/percent at 14:45:
93.37
QoS: 0 - Retain: false
Message 14 received on papa/modemsleep/get at 14:45:
300000
QoS: 0 - Retain: false
Message 13 received on papa/humidity/measured/get at 14:45:
51.73
QoS: 0 - Retain: false
Message 12 received on papa/temperature/measured/get at 14:45:
19.84
QoS: 0 - Retain: false
Message 11 received on papa/log at 14:45:
[INFO]  [0000917576] MQTT subscribed to topic: papa/debug/loglevel
QoS: 0 - Retain: false
Message 10 received on papa/log at 14:45:
[INFO]  [0000917570] MQTT subscribed to topic: papa/temperature/target/set
QoS: 0 - Retain: false
Message 9 received on papa/log at 14:45:
[INFO]  [0000917554] MQTT subscribed to topic: papa/modemsleep/set
QoS: 0 - Retain: false
Message 8 received on papa/log at 14:45:
[INFO]  [0000917545] MQTT subscribed to topic: papa/mode/set
QoS: 0 - Retain: false
Message 7 received on papa/log at 14:45:
[INFO]  [0000917330] MQTT subscribed to topic: papa/debug/enable
QoS: 0 - Retain: false
Message 6 received on papa/log at 14:45:
[INFO]  [0000917324] MQTT topic: papa/, topic len: 5
QoS: 0 - Retain: false
Message 5 received on papa/log at 14:45:
[INFO]  [0000917310] Device startup and setup done
QoS: 0 - Retain: false
```

alexmohr, 2021-11-21:
The target temperature only will be updated if something is set in the mqtt topic. As you restarted everything probably no retain message is set in `papa/temperature/target/set` 
If you subscribe to this topic a message with the set temperature should always be received.
If no message is set, the target temperature won't be confirmed by the esp and thus nothing will be sent and home assistant will stay empty. 

So an easy fix for this is just set the temperature in home assistant.
I'm not so keen of sending a default because it would make the mqtt logic much more complicated, as the receiving of messages is done asynchronous to the rest of the logic and might be executed after mqtt::loop is done.

Anycubic, 2021-11-21:
I did set a temperature from HA interface but still the issue. I never had this issue before, it popped up with latest web config new version

alexmohr, 2021-11-21:
Could you post what you receive in the `papa/temperature/target/set` topic?

Anycubic, 2021-11-21:
I switched to 21 and nothing (what you see at 15:38 is the very first log start, 21 was set 30 minutes before). Then switched back to 20 and it seems that this time it worked, now I can see the set temperature also in HA interface. Then switched back to 21:

```
Message 41 received on papa/temperature/target/set at 15:48:
21
QoS: 0 - Retain: false
Message 40 received on papa/temperature/target/set at 15:39:
20
QoS: 0 - Retain: false
Message 39 received on papa/temperature/target/set at 15:38:
21
QoS: 0 - Retain: true
```

I can't understand if there is a logic

alexmohr, 2021-11-21:
The first two messages are wrong. They are sent without retain flag. (`Retain: false`). Without retain flag only clients connected to the broker receive the message. 
It is vital that the retain flag is set, so the esp doesn't has to be online at the exact moment the message is published. With retain flag it can connect at any time and will receive the last retained message.

closed 2021-11-21 by alexmohr

Anycubic, 2021-11-21:
So there must be a bug in HA. There is the `retain: true` flag in the configuration...

alexmohr, 2021-11-21:
I took a look at the home assistant code and it looks correct and hasn't changed in this part for the last 2 years. It's quite odd that it doesn't send the retain flag for the first two messages but works afterwards

Anycubic, 2021-11-21:
Were you able to reproduce it?

alexmohr, 2021-11-22:
I haven't tried yet