alexmohr/open-heat #28: STL file disappeared

Anycubic, 2021-11-22:
Hi, I'm working on a custom case STL. Could you please publish again the STL file? There are just Freecad files in Schematics folder but unfortunately Freecad crashes on startup on my Mac.

alexmohr, 2021-11-23:
I can't publish STL files anymore as they are over 200mb large since I've added threads for the screw holes. GitHub does not allow auch large files.

Anycubic, 2021-11-23:
Could you please zip it and share it to me? I've lost all of the old builds file and it seems to me I can't download it anymore

alexmohr, 2021-11-23:
Not for the moment. The case is not tested yet and therefore I have no intention whatsoever to support it in anyway before getting it printed and trying it out.

Anycubic, 2021-11-23:
I'm not going to use it "as is", I just need to take the part where it holds the valve. Also the old STL version will do the job

Anycubic, 2021-11-23:
Nevermind, I just found an old zip with old STLs

closed 2021-11-23 by Anycubic

alexmohr, 2021-11-23:
Alright but be warned, I'm not sure if this works at all. That's why I'm very hesitant to share these untested things. 
If you design a custom case and probably better case it would be great if you would share it.

Anycubic, 2021-11-23:
> Alright but be warned, I'm not sure if this works at all. That's why I'm very hesitant to share these untested things. If you design a custom case and probably better case it would be great if you would share it.

Of course I will share my work!!!