alexmohr/open-heat #29: Latest Dev release went to config mode by itself with ESP-12F

Anycubic, 2021-11-23:
One hour ago the ESP-12F stopped posting, to my surprise it was in config mode! I don't know if it is related, but double reset does NOT make the ESP-12F in config mode, while more than once I saw in terminal that after a single reboot it was going in config mode. Did you fix NodeMCU and broke ESP-12F?

alexmohr, 2021-11-23:
Must be broken in the library then as I removed the special handling for esp12 ...
Probably won't fix this

alexmohr, 2021-11-23:
As for the spontaneous config mode: I can't do anything without logs.
Eventually I'll go back to the esp and fix it if I can reproduce it.

Anycubic, 2021-11-23:
That's too bad. I would remove totally the double reset thing, I never used it, just enable log -> web config -> flash -> debug off again -> prosit!
Also in production environment it is totally useless, you should solder a reset button, take the valve out, etc etc

alexmohr, 2021-11-23:
As soon as a serial adapter is connected double reset detection is working fine on esp 12 :(
But it seems that you don't care about the drd but rather the config mode. 

When this happens again you can try to remove the double reset detector in main. 
But I don't think that's the issue. 
Maybe your wifi network was gone when the ESP woke up. It goes into config mode if it can't connect to wifi. 
There is a timeout of 6s and after that it goes into config mode. 

If possible, please provide logs without them it's near impossible to troubleshoot anything. 
Thanks

Feel free to re-open this issue then

alexmohr, 2021-11-23:
Oh and I removed a ton of delays from the WiFi setup to speed up the connect time and reduce the time is on.

If you have this issue again I'll add a connect retry if the device woke up from deep sleep.

Thanks again 👍

Anycubic, 2021-11-23:
> There is a timeout of 6s and after that it goes into config mode.

Maybe that's the culprit, that valve hasn't a strong WIFI signal

> If you have this issue again I'll add a connect retry if the device woke up from deep sleep.

I would implement that, the downside could be that the radiator goes offline...

alexmohr, 2021-11-24:
If there is no WiFi signal there's not really anything I can do. Maybe fully close the valve to prevent over heating

alexmohr, 2021-11-29:
I'm fairly certain it is WiFi related due to the implementation of the double reset detector. 
Therefore I won't remove it for now.
https://github.com/alexmohr/open-heat/blob/old_regulator/src/main.cpp#L122 replace this implementation with `return false` to disable it ;)

Anycubic, 2021-11-29:
Very good, I'll give it a try, thank you!

Anycubic, 2021-11-29:
> I'm fairly certain it is WiFi related

You could be right (as always) but it should not be a matter of WIFI signal strength, maybe my WIFI router is just clogged.with so many WIFI client popping out

Anycubic, 2021-11-29:
Where do I change those 6 seconds WIFI connection waiting time? Maybe changing to 10 would help? I had a brief look at your code but for tonight I surrender. Thank you

alexmohr, 2021-11-30:
https://github.com/alexmohr/open-heat/blob/old_regulator/src/network/WifiManager.cpp#L154 replace 60 with 100

Anycubic, 2021-11-30:
And we have a winner:

```
[INFO]  [0002132743] Connecting WiFi...
[DEBUG] [0002134927] Using fast connect
[WARN]  [0002141776] WiFi connect timeout
```

I don't know if this ESP is faulty, it is literally 2/3 meters away from my WIFI router. 
I learnt PCB position is important for WIFI sensitivity, maybe also locking the WIFI module in b mode could help:

[image]

alexmohr, 2021-11-30:
I'm using a 12-E which should have _worse_ wifi than the F variant as this had some reworks to wifi and antenna design etc. Maybe it's a combination between router and esp. 
I suspect that the fast connect path is not working reliable for your setup. 
You can try to disable the fast connect mode. 
https://github.com/alexmohr/open-heat/blob/dev/src/network/WifiManager.cpp#L140
just set `hasFastConfig` to false or remove the block from the if. 

I don't want to implement a fixed wifi mode.

Anycubic, 2021-11-30:
> I don't want to implement a fixed wifi mode.

I can give it a try. Where do I need to add the relevant code? 

Closing this as not code related

closed 2021-11-30 by Anycubic

alexmohr, 2021-11-30:
> > I don't want to implement a fixed wifi mode.
> 
> I can give it a try. Where do I need to add the relevant code?
> 
> Closing this as not code related

I'd try to remove the fast connect first if I'd be you. I don't know how to set a fixed wifi mode. You'd have to google that yourself

Anycubic, 2021-11-30:
Yes and yes. I just wanted to know in which part of your code I need to put my hands, I'm confused since there is the web config part and the "work as usual" part

alexmohr, 2021-11-30:
I already linked the connect method above. WiFi manager is always used to provide ..  well WiFi ... 
It's for the dev branch but you'll manage ;)

Anycubic, 2021-11-30:
>  I suspect that the fast connect path is not working reliable for your setup. You can try to disable the fast connect mode. 

It seems this was it, time will tell if this is indeed true

Anycubic, 2021-12-01:
It seems that this is going to be useful: `WiFi.setPhyMode(WIFI_PHY_MODE_11B);`

Anycubic, 2021-12-03:
I didn't have any issue anymore, forcing mode b only solved the fault connection problem