alexmohr/open-heat #31: Battery values are pushed to MQTT erratically

Anycubic, 2021-11-25:
I'm working with last old-regulator build only. I noticed that battery values are always pushed when I switch on the valve, after that they are not pushed to MQTT every modem wake up but very erratically, while GET measured temp is very steady. Did you noticed that?

alexmohr, 2021-11-25:
I don't believe this ... 
https://github.com/alexmohr/open-heat/blob/old_regulator/src/network/MQTT.cpp#L105
How did you check this?
And please _always_ attach logs if you open an issue
Logs from today:
```
08:36:50.61	00:14:41
08:51:47.34	00:14:57
09:06:44.04	00:14:57
09:21:41.49	00:14:57
09:36:39.46	00:14:58
09:51:35.91	00:14:56
10:06:31.81	00:14:56
10:21:28.83	00:14:57
10:36:27.14	00:14:58
10:51:23.60	00:14:56
11:06:21.39	00:14:58
11:21:17.60	00:14:56
11:36:13.59	00:14:56
11:51:10.80	00:14:57
12:06:06.86	00:14:56
12:20:59.83	00:14:53
12:35:58.39	00:14:59
12:50:55.80	00:14:57
13:50:40.88	00:59:45
14:05:38.65	00:14:58
14:20:35.26	00:14:57
14:35:31.26	00:14:56
14:50:24.69	00:14:53
```

raw output
```
08:22:09.973 > [DEBUG] [0000004107] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
08:36:50.605 > [DEBUG] [0000908288] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
08:51:47.342 > [DEBUG] [0001812465] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
09:06:44.043 > [DEBUG] [0002716658] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
09:21:41.490 > [DEBUG] [0003621962] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
09:36:39.462 > [DEBUG] [0004526114] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
09:51:35.914 > [DEBUG] [0005430258] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
10:06:31.814 > [DEBUG] [0006334414] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
10:21:28.829 > [DEBUG] [0007238561] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
10:36:27.135 > [DEBUG] [0008142715] MQTT send '3.39' in topic 'living_room/heater_bed/battery/voltage'
10:51:23.601 > [DEBUG] [0009046880] MQTT send '3.39' in topic 'living_room/heater_bed/battery/voltage'
11:06:21.391 > [DEBUG] [0009951070] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
11:21:17.602 > [DEBUG] [0010855268] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
11:36:13.593 > [DEBUG] [0011759478] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
11:51:10.801 > [DEBUG] [0012663667] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
12:06:06.859 > [DEBUG] [0013567810] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
12:20:59.832 > [DEBUG] [0014471962] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
12:35:58.392 > [DEBUG] [0015376115] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
12:50:55.804 > [DEBUG] [0016280256] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
13:50:40.879 > [DEBUG] [0019896956] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
14:05:38.648 > [DEBUG] [0020801149] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
14:20:35.256 > [DEBUG] [0021705335] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
14:35:31.264 > [DEBUG] [0022609482] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
14:50:24.694 > [DEBUG] [0023513625] MQTT send '3.42' in topic 'living_room/heater_bed/battery/voltage'
```

And please always attach logs when opening bugs

alexmohr, 2021-11-25:
The 1 one in between is due to the device beeing turned off

Anycubic, 2021-11-25:
I'm using MQTT Explorer, as you can see measured temps are captured in a steady way, battery% (and voltage) only 3 times in the same timeframe and I can't understand why.
[image]
[image]

alexmohr, 2021-11-25:
Take a look in the raw mqtt traffic and I'm almost certain you'll see every single message. 
This tool probably only logs a new point when it's different from the old one. 
Temperature is changing a lot even if it's only by .01°C while battery is not changing that often.
For example grafana is doing the same thing. 

Closing this as out of scope ...

closed 2021-11-25 by alexmohr

Anycubic, 2021-11-25:
Ok, I will have a better look into it, I'll never stop learning from you ;-)

alexmohr, 2021-11-25:
Just use mosquitto if you don't believe me :wink: 
`mosquitto_sub -h your_broker_host -t "battery_topic" `

Anycubic, 2021-11-25:
Are you joking? I believe in you more than in myself! :-D
That's why I will follow your suggestion.... ha ha ha :-D

Anycubic, 2021-11-25:
WTF!!! I'll try to reflash `papa` ....
[image]
[image]

alexmohr, 2021-11-25:
Check your wiring too ;)

Anycubic, 2021-11-25:
Could it be poor WIFI signal? But how it comes the temps are transmitted at every wake up and not battery values AT ALL? So no "0" or wrong values, just nothing. I reflashed via OTA but the issue is still there. Is there anything in memory surviving an OTA flash + power cycling?

alexmohr, 2021-11-25:
The screenshot you posted show battery levels? What issue do you mean?

alexmohr, 2021-11-25:
The configuration is kept after flashing. 
I have no idea whats wrong and I can't help you until I see logs

Anycubic, 2021-11-25:
I'll deploy other valves and will try to understand, by now I'd like to focus on going in production and test how robust home assistant + MQTT + your code is ;-). Btw the problem should not be in the code

Anycubic, 2021-11-26:
So I'm serial logging an ESP without sensors. In say 11 sleep/wakeup cycle I had this error:

```
[17:57:41][INFO]  [0000620691] Wifi connected:
[17:57:41]	time: 3229
[17:57:41]	SSID: Mirabello_NG
[17:57:41]	RSSI=-44
[17:57:41]	Channel: 11
[17:57:41]	IP address: 192.168.2.209
[17:57:41][INFO]  [0000620692] Running MQTT setup
[17:57:41][INFO]  [0000620707] Device startup and setup done
[17:57:41][INFO]  [0000621054] MQTT topic: ProvaNG/, topic len: 8
[17:57:41][INFO]  [0000621137] MQTT subscribed to topic: ProvaNG/debug/enable
[17:57:41][INFO]  [0000621205] MQTT subscribed to topic: ProvaNG/mode/set
[17:57:41][INFO]  [0000621278] MQTT subscribed to topic: ProvaNG/modemsleep/set
[17:57:41][INFO]  [0000621348] MQTT subscribed to topic: ProvaNG/temperature/target/set
[17:57:41][INFO]  [0000621412] MQTT subscribed to topic: ProvaNG/debug/loglevel
[17:57:41][DEBUG] [0000621417] MQTT send '0.00' in topic 'ProvaNG/temperature/measured/get'
[17:57:41][DEBUG] [0000621418] MQTT send '300000' in topic 'ProvaNG/modemsleep/get'
[17:57:41][DEBUG] [0000621460] MQTT send '0.00' in topic 'ProvaNG/battery/percent'
[17:57:41][DEBUG] [0000621462] MQTT send '0.94' in topic 'ProvaNG/battery/voltage'
[17:57:41][DEBUG] [0000621465] Sleep times: valveSleep: 18446744073708930150, mqttSleep: 299999
[18:02:25][INFO]  [0000621522] Sleeping for 299992 msrl[18:02:25][DEBUG] [0000921597] FS setup done
[18:02:25][DEBUG] [0000921605] Loading config
[18:02:25][DEBUG] [0000921606] Successfully loaded config
[18:02:25][DEBUG] [0000921606] woke up from deep sleep
[18:02:25][DEBUG] [0000921608] Set last reset time to 921608
[18:02:25][DEBUG] [0000921613] Running setupPins
[18:02:25][DEBUG] [0000921617] Running setupTemperatureSensor
[18:02:25][INFO]  [0000921622] Setting up BMP280
[18:02:25][INFO]  [0000921627] BMSensor init result: 0, try: 1
[18:02:25][INFO]  [0000921732] BMSensor init result: 0, try: 2
[18:02:25][INFO]  [0000921832] BMSensor init result: 0, try: 3
[18:02:25][INFO]  [0000921933] BMSensor init result: 0, try: 4
[18:02:25][INFO]  [0000922033] BMSensor init result: 0, try: 5
[18:02:28][ERROR] [0000924634] Failed to init temp sensor
[18:02:28][DEBUG] [0000924635] no double reset, resetInfo: Deep-Sleep Wake
[18:02:28][INFO]  [0000924638] Connecting WiFi...
[18:02:30][DEBUG] [0000926822] Using fast connect
[18:02:31][INFO]  [0000927869] Wifi connected:
[18:02:31]	time: 3229
[18:02:31]	SSID: Mirabello_NG
[18:02:31]	RSSI=-43
[18:02:31]	Channel: 11
[18:02:31]	IP address: 192.168.2.209
[18:02:31][INFO]  [0000927870] Running MQTT setup
[18:02:31][INFO]  [0000927885] Device startup and setup done
[18:02:32][INFO]  [0000928225] MQTT topic: ProvaNG/, topic len: 8
[18:02:32][INFO]  [0000928298] MQTT subscribed to topic: ProvaNG/debug/enable
[18:02:33][ERROR] [0000929600] MQTT failed to subscribe to topic: ProvaNG/mode/set
[18:02:33][ERROR] [0000929601] MQTT last error: -9
[18:02:33][ERROR] [0000929601] MQTT failed to subscribe to topic: ProvaNG/modemsleep/set
[18:02:33][ERROR] [0000929609] MQTT last error: -9
[18:02:33][ERROR] [0000929613] MQTT failed to subscribe to topic: ProvaNG/temperature/target/set
[18:02:33][ERROR] [0000929621] MQTT last error: -9
[18:02:33][ERROR] [0000929625] MQTT failed to subscribe to topic: ProvaNG/debug/loglevel
[18:02:33][ERROR] [0000929633] MQTT last error: -9
[18:02:33][DEBUG] [0000929639] MQTT send '0.00' in topic 'ProvaNG/temperature/measured/get'
[18:02:33][ERROR] [0000929645] Mqtt publish failed: -9
[18:02:33][DEBUG] [0000929650] MQTT send '300000' in topic 'ProvaNG/modemsleep/get'
[18:02:33][ERROR] [0000929657] Mqtt publish failed: -9
[18:02:33][DEBUG] [0000929697] MQTT send '0.00' in topic 'ProvaNG/battery/percent'
[18:02:33][ERROR] [0000929698] Mqtt publish failed: -9
[18:02:33][DEBUG] [0000929699] MQTT send '0.99' in topic 'ProvaNG/battery/voltage'
[18:02:33][ERROR] [0000929706] Mqtt publish failed: -9
[18:02:33][DEBUG] [0000929711] Sleep times: valveSleep: 18446744073708621904, mqttSleep: 299999
[18:07:16][INFO]  [0000929768] Sleeping for 299992 msrl[18:07:16][DEBUG] [0001229842] FS setup done
[18:07:16][DEBUG] [0001229850] Loading config
[18:07:16][DEBUG] [0001229851] Successfully loaded config
[18:07:16][DEBUG] [0001229851] woke up from deep sleep
[... 10 lines cut ...]
[18:07:19][DEBUG] [0001232880] no double reset, resetInfo: Deep-Sleep Wake
[18:07:19][INFO]  [0001232883] Connecting WiFi...
[18:07:21][DEBUG] [0001235067] Using fast connect
[18:07:22][INFO]  [0001236114] Wifi connected:
[18:07:22]	time: 3229
[18:07:22]	SSID: Mirabello_NG
[18:07:22]	RSSI=-41
[18:07:22]	Channel: 11
[18:07:22]	IP address: 192.168.2.209
[18:07:22][INFO]  [0001236115] Running MQTT setup
[18:07:22][INFO]  [0001236131] Device startup and setup done
[18:07:23][INFO]  [0001236499] MQTT topic: ProvaNG/, topic len: 8
[18:07:23][INFO]  [0001236570] MQTT subscribed to topic: ProvaNG/debug/enable
[18:07:23][INFO]  [0001236638] MQTT subscribed to topic: ProvaNG/mode/set
[18:07:23][INFO]  [0001236707] MQTT subscribed to topic: ProvaNG/modemsleep/set
[18:07:23][INFO]  [0001236779] MQTT subscribed to topic: ProvaNG/temperature/target/set
[18:07:23][INFO]  [0001236846] MQTT subscribed to topic: ProvaNG/debug/loglevel
[18:07:23][DEBUG] [0001236851] MQTT send '0.00' in topic 'ProvaNG/temperature/measured/get'
[18:07:23][DEBUG] [0001236852] MQTT send '300000' in topic 'ProvaNG/modemsleep/get'
[18:07:23][DEBUG] [0001236894] MQTT send '0.00' in topic 'ProvaNG/battery/percent'
[18:07:23][DEBUG] [0001236895] MQTT send '0.98' in topic 'ProvaNG/battery/voltage'
[18:07:23][DEBUG] [0001236899] Sleep times: valveSleep: 18446744073708314717, mqttSleep: 299999
[18:12:06][INFO]  [0001236956] Sleeping for 299992 msrl[18:12:06][DEBUG] [0001537029] FS setup done
```
Is this `MQTT failed to subscribe to topic` just something which happens sometime?

alexmohr, 2021-11-26:
Shouldn't happen and indicates invalid configuration of some sort. But your setup is not working anyways as the temperature sensor couldn't be set up either

Anycubic, 2021-11-26:
I guess it was a sort of glitch, btw I'm going to plug a full configured ESP to that Windows XP laptop