alexmohr/open-heat #32: Valve in bad positions: ideas to deal with it

Anycubic, 2021-11-25:
In general having a temp sensor close to the very same radiator is not a good idea...but I have several high sized radiators where the valve is at approx 2,5 meters from the floor level.... so there is a big delta, even 2 Celsius and more, between temps at floor level and where the valves are in those high radiators.
I was thinking that a possible solution could be:

1) moving the BMP/BME away and in lower position from the valve. Don't know if long wires could be a problem, also aesthetically not very nice solution
2) having a separate ESP/BMP sensor only fixed in a nicer position and pushing measured temps only (so the valve will take them via MQTT instead that from internal sensor). Adding a display and a potentiometer this could also became a real "room" thermostat (though I guess you need wire tap, battery will not be enough because of display...)
3) very quick & dirt, having the possibility to add a correction value (via web interface) to the measured temps so that it can be possible to deal with these problematic  radiators like if they are normal position (i.e. I don't need to remember I have to set 2 C more than usual). In general this could only fix "problematic" sensors, i.e. I have a BME reading 1,5 C more than real temp.

What do you think? Thank you

alexmohr, 2021-11-25:
You would have these issue with most commercial solutions as well: 

1. I2C is designed for chip to chip communication on the same board, so long wires might not work. I tested 0.5 metres with twisted pair wires and it worked but no guarantee on that.
2. This will require to turn on the WiFi every 5 minutes to fetch the new temperature and will drain the battery significantly. It would be supprted by the software architecture but I'm not going to implement this any time soon.
3. Just correct the temperature in home assistant by selecting it higher or lower o don't think I have to handle this in firmware.

So I would say go for option 3. If you want something nice option 2 would be the way to go.

Anycubic, 2021-11-25:
Indeed I arrived to the same conclusion, thanks

closed 2021-11-25 by Anycubic